martinloren/HScope #4: 5v Best Module IC

petanque, 2019-03-29:
Hi, I wonder why you use IC AMS1117-1.2v in 5v Best Module.
The output pin is not used, is 1.2v important or can it be replaced with a 1.8v or 3.3v?

martinloren, 2019-04-02:
Ive tested ASM1117-ADJ version amd it work as well. This configuration is not documented and provide VCC/2 required to shift the input signal in the middle of the ADC range. In this way you can have a dual input without having a dual voltage ADC stage normally required for these devices. This allow the less number of components possible.

MikkelSnitker, 2019-06-13:
Hi MartinLoren.

So the output of the ASM1117 is not connected to anything? 

Since my electronics pusher doesn't have the ASM1117, I was wondering if you think the TS1117-ADJ  [(http://www.farnell.com/datasheets/50383.pdf)](http://www.farnell.com/datasheets/50383.pdf) would do as well?

martinloren, 2020-01-11:
Yes, output not connected. For other components you should test. I've tested also another small version of ASM1117 and it is not working, it is not the same circuit.

mmghv, 2021-07-08:
I honestly don't know how it's working and I need help, I don't have access to an ASM1117 but I have other regulators, does it have to be 1.2v or the voltage doesn't matter here since the output is not used, how can I test to see if my regulator is working as expected?

I'm very confused about the use of this ASM1117 in the circuit and I never seen it being used like this before.