martinloren/HScope #17: Device not recognised on Xiaomi A1

raomin, 2019-08-09:
Hi @martinloren,

I get "Device not connected" on my phone Xiaomi A1. Maybe I'm doing something wrong:
- I flashed the firmware on the STM32 (tried v4 and v5). It flashed ok
- Then I've reset the jumper, removed the Serial connection and plugged it to the phone with the otg
- The leds turns fix green immediately 
- I open the app, it shows "Device not connected"

When I connect the stm32 on a windows machine, no device get recognised...

Did I miss something or is there a problem with my phone?
How can I debug this?

Thanks!

raomin, 2019-08-16:
Hi @martinloren, I tried on a Sony phone yesterday. Same issue....
Any help on this?

martinloren, 2019-08-24:
Hi I've found some bug in the Firmware, now fixed in version 5. Please try new version 5. Also HScope app use the Beta one.

I look forward for your tests.

raomin, 2019-09-02:
Hi @martinloren,
Your fix made it! It is detected and shows a signal.
I'll try with the module I prepared and close the ticket if all goes well.

Thanks!

raomin, 2019-09-03:
All is fine with the module. Thanks for your support.
BTW do you plan to release the code of the firmware?

closed 2019-09-03 by raomin

martinloren, 2019-09-18:
No sorry, I'm not going to release the firmware since this is not an open source project.
Enjoy!