martinloren/HScope #18: Sept 2019 Improvements

martinloren, 2019-09-16:
**Zero Level Calibration**
The pop up window says:
2) Sconnect the probe or shortcut to GND. This should probably read 'Disconnect'. Maybe it is even better to leave disconnect out and always advise to connect to GND.
 
 
**Positioning channel 2: green arrow is hidden**
When activating channel 2, the green arrow to change position is hidden behind the sidebar tab. It is hard to get hold of it. See image ‘Channel2_Positioning_1’.
Possible solution:
Move the sidebar tab further to the top.
Or, don’t have a sidebar tab at all, sidebar is opened and closed by swiping.
 
 
**Positioning channel 2: can’t grab the green arrow when it is at the bottom**
I somehow managed to get the situation like in image ‘Channel2_Positioning_2’.
I was not able to grab the green arrow to move the signal … adding a purple cursor has priority.
 
 
**Single trigger & Unlock: looks at level in stead of edge**
See images ‘Trigger1’ and ‘Trigger2’. Single trigger is activated, channel2, upward edge. I hoped that pressing 'Unlock' would start display when the signal has its first upward edge. What happens however is when the signal already is higher than the trigger level, display starts immediately … rendering only part of the first pulse.
Possible solution:
Looks like acquisition is activated by: IF sample > trigger THEN start.
Better would be: IF previous_sample <= trigger AND current_sample > trigger THEN start.
 
It also leads to the display of signal values that were not there … which is even worse when Phosphor mode is switched on … see next point.
 
 
**Trigger Unlock > scale changes, strange phosphor effects**
If Single trigger & Unlock is used, then initially the scale changes and shortly after the scale goes to the set values. This leads to a short screen flicker after Unlock. Moreover … when Phosphor effect is active, sometimes an artifact stays visible in this situation. See image Phosphor_Unlock_Scale.
 
 
**Pan & zoom**
When the 'Scope' button is switched off and acquisition stops, initially the display can not be dragged vertically, only horizontally. Only after a two finger  pinch, or after tapping the 'Zoom to Fit' icon, the screen can be dragged vertically.
 
 
**Scale on the voltage meter is misaligned.**
See image ‘Voltage_Scale_Misaligned’.
 
 
**Demo signals are not static**
The demo signals change when time scale or amplitude scale is changed. This is rather unexpected and trying out the effects of scale changes in demo mode is rather impossible.
 
 
 
**NICE TO HAVE**
 
**Horizontal (voltage) cursors.**
 
**Channel1 arrow**
Only channel2 has an arrow cursor. It would be nice if channel1 also has an arrow.
 
 
**Trigger adjustment to a specific value**
The red trigger cursor arrow is hard to position at an exact value. Say I want to set it at exactly 2.0V, or later back to 0.0V … it is very hard to do.
Possible solutions:
- a number entry field
- snap to grid, similar like Channel2 position cursor
- some kind of zoom
- one finger drag = course, two finger drag = fine
- a range slider like with the Android app called Function Generator’, which is very fine in the middle and course at the edges.
 
The same holds for the time scale cursors … it is difficult to position them at an exact sample.
 
 
**Display of the sample ranges / frequencies**
The character used for kilo is k … non capital.
 
A space between the number and the units will improve readability: 10 kHz in stead of 10KHz.

**Mearurement results display position fluctuations**
When the Measurements button is pressed the measurement results show in the neighborhood of the actual signal. While this can be handy to distinguish them with 2 channels displayed, the backside is that their position fluctuates when the signal fluctuates. An alternative cold be to have two boxes, one yellowish, one greenish, stable positioned, with the measurement results displayed in them.

**Settings page.**
When tapping settings, why not by default start in the General Settings in stead of Info?
The number of chapters is short enough to all be on the top level, in stead of having paragraphs under the Settings chapter.
The font could be larger ... I have a hard time to tap the correct item on my tablet.

**Zoom resets after tapping 'Scope' button.**
Starting and stopping acquisition by tapping the 'Scope' button changes the display back to 'no zoom', also after the user has just zoomed in manually, or has tapped the 'Zoom to Fit' icon. Can it stay at the current zoom setting when tapping 'Scope'?

**'Zoom to Fit' icon only zooms vertical**
When tapping the 'Zoom to Fit' icon, only the vertical, voltage, zoom is applied. Why not also change the horizontal zoom such that, at least for a repetitious signal where a frequency can be measured, 3 periods are in view and T0 is in the middle?
Actually ... why not always automatically zoom to fit? Unless the user has zoomed by hand to his liking, then keep that zoom setting.

**'Auto' button.**
The 2  channels have an auto scale button. Could there also be an overall 'Auto' button, after which Hscope analyzes the incoming signal and automatically selects the best time scale and voltage scale?

RudyB24, 2019-09-16:
Martin, the version on my phone and tablet is 136. (I do not see it being marked beta.)

martinloren, 2019-09-17:
Google Tests in progress, it should be available soon...

RudyB24, 2019-09-20:
Test findings Hscope version 2.2.1 (137)

Only mentioning issues that are not (completely) solved / changed

**Single trigger & Unlock: looks at level in stead of edge.**
Still does not trigger at the rising edge, but at the level the signal had when Unlock was tapped. 
See [this image](https://drive.google.com/open?id=11I8aeBxZuxDkoOJLa-gjS6CVqdEQV0az).

Single trigger dirty samples at the beginning 
Not yet entirely gone ... see [this image](https://drive.google.com/open?id=11I8aeBxZuxDkoOJLa-gjS6CVqdEQV0az).

**Pan & zoom**
When the 'Scope' button is switched off and acquisition stops, initially the display can not be dragged vertically, only horizontally. Only after a two finger pinch, or after tapping the 'Zoom to Fit' icon, the screen can be dragged vertically.

--> Cannot be replicated, provide sequence of buttons used to arrive to this condition. Probably the Y screen is already zoomed to maximum.

Correct, when started at minimal zoom, scale -6 <> +6V vertical positioning is 'locked'. with any other scale zoom can start immediately. 
**Question:** would it be possible to zoom out anyhow, making the display scale larger than 6V? Then also the signal can be dragged to any position.

**Scale on the voltage meter is misaligned.**
Is still misaligned, [see image](https://drive.google.com/open?id=11Qfj6rhpnpWhp0b-ukYuJQta8KYJ9TJh).

**Settings page.**
Improved, but fonts could still be larger. It could be more clear where to click. [See image](https://drive.google.com/open?id=11hxQFQwNY6H-N-ZG_REFMD7v_Q1yTpDQ).

If for example I want to change Samples interpolation, I can't click that word, I need to click Linear, at the far right. With a thick finger I can easily click the line above or below. Also the drop down elements are still small and error prone. A larger font or more space between the lines can help. Or ... since the number of choices in the drop down table is never more than 4, maybe it could be radio buttons or tick boxes which are always in view? 

**'Zoom to Fit' icon only zooms vertical**

--> In most of cases you want to navigate on a custom zoom in X and enlarge to fit just along Y. Fit according X has no sense since you can have 2 signals at difference frequency, so which one you would like to fit automatically?
Either auto zoom to show 3 periods of the fastest of the two signals. Or auto zoom to channel1 (less effort to build I bet).

**'Auto' button.**
The 2 channels have an auto scale button. Could there also be an overall 'Auto' button, after which Hscope analyzes the incoming signal and automatically selects the best time scale and voltage scale?

--> This is a cool feature to consider...
Same question holds here. Zoom to the fastest signal, or zoom to channel1 (less effort to build I bet).

AlfaBravoX, 2022-02-20:
> 'Auto' button.
The 2 channels have an auto scale button. Could there also be an overall 'Auto' button, after which Hscope analyzes the incoming signal and automatically selects the best time scale and voltage scale? 
This is a cool feature to consider...
Same question holds here. Create setting and zoom for the fastest signal, or find settings and zoom to channel1 (less effort to build I bet).

Has anything been done on 'Auto' button feature? On IDSO1070A stock app delivered by hantek, this is included and I am heavily using this. It is not pleasant to switch between 2 apps, just to use AUTO at stock one and the goto back to Hscope.