martinloren/HScope #21: Request for HS104

zx080, 2019-12-02:
First of all, I am sorry for opening an issue entry but actually I could not find any other practical method.

Dear martinloren, thank you for your wonderful work. It would be very practical using STM32 as a DSO device. Also, it would be very very nice a four-channel version of HS10x series. 
I have a request for HS104 (four-channel version).
Thanks in advance
Regards

martinloren, 2019-12-07:
Interesting request, I would like to implement it once HScope will support 4 channels :D

martinloren, 2020-06-19:
After having developed HS402, a 4 channels may be implemented with a specific HS404 based on STM32F407 or directly with the HS704 (based on STM32H750)

warehouseElc, 2022-09-08:
how do i get 4 channel hardware for hs402 ?