martinloren/HScope #33: BUG: HS102 Pro Channel 2 AC/DC selection on reconnect

gadjex, 2020-06-15:
On an HS102 Pro, after turning on channel 2 to DC mode and then disconnecting the USB and reconnecting, the app will erroneously show DC mode on channel 2 when it is actually in AC mode. Switching channel 2 off and then back to DC mode will fix the issue until the next disconnect and 
reconnect.

This is confirmed by using a volt meter and checking pin PB13 is at 0 volts after reconnecting.

Channel 1 does not display this improper behavior.

martinloren, 2020-06-19:
Thanks for reporting this issue. What is the firmware version?

gadjex, 2020-06-19:
Happens with v9 and v10.1

martinloren, 2020-06-19:
Confirmed, it is a bug on HScope, it will be fixed in next Beta release. By the way there is no HS102 Pro....did you build it? I mean with the PGA and all range of input voltages or just single voltage range?

gadjex, 2020-06-19:
I built it with 2 PGAs 2 SSRs and 2 AMS1117. Both channels work +-15V. I
haven't fully tested it. I can send you the schematic when I get home
tomorrow. It really is just a HS101 Pro with a duplicate channel.