martinloren/HScope #36: Problems with hantek 1008C not recognised.

mazzerman, 2020-08-03:
Hi. I have the paid full vesion with automotive module. I bought an Android pad with OTG compatibility. (Tested with OTG check app). I also bought OTG compatible USB C connector. I launch Hscope but it will only show DEMO mode. It says device not connected. I have the Hantek powered separately. Are there any drivers I need to install alongside the Hscope app?
Regards

mazzerman, 2020-08-12:
I have an android pad that is confirmed OTG comparable. I have also now purchased 3 different OTG connectors 2 small cable style with mini C USB to standard USB  and the software still stays in demo mode can someone please help because I've had this software for a few months and never been able to use it.
[image]

mazzerman, 2020-08-12:
[image]
[image]
[image]
[image]
[image]
[image]

martinloren, 2020-10-07:
It may depend from the tablet, try any other phone, can also ask a friend to try with his phone. Check also the testing methods here: https://www.martinloren.com/hscope/how-can-we-help-you/

closed 2020-10-07 by martinloren

mazzerman, 2020-10-07:
Hi I cannot get Hscope to work with my Lenovo android pad.
The only USB options are on the attached photo. That is with it plugged into the laptop.
With the OTG cohort cable it greys out.[Image]